steveohara/j2mod #3: j2mod on Android

djadry, 2016-04-19:
Hello, I'm triyng to use j2mod on an Android application on RTU, but I saw that the library needs the jSerialComm dependency that requires /dev/ttix device. So it does't work with a specific kernel drive.
I am planning to use https://github.com/felHR85/UsbSerial for the low level communication.
Do you have any advice about this?

steveohara, 2016-04-19:
That looks like an interesting way to get UART capability through USB.
Do you have any experience of using that library? I'm thinking particularly in terms of stability and QA.
In any event, I would be happy to add it to j2mod as we could use it here at 4energy

djadry, 2016-04-20:
Now I'm trying to implement UsbSerial istead of jSerialComm for the serial low level commnication...

cusna, 2016-06-19:
I too am very interested to use the library j2mod through USB with a converter(type ftdi or prolific)
serial. It would be interesting to incorporate in j2mod the library [https://github.com/felHR85/UsbSerial](https://github.com/felHR85/UsbSerial) or [https://github.com/mik3y/usb-serial-for-android](https://github.com/mik3y/usb-serial-for-android).

steveohara, 2016-08-20:
This looks like it has been resolved by https://github.com/steveohara/j2mod/pull/28 so I'm closing the ticket in the expectation that this works fine and will be available in an upcoming release

closed 2016-08-20 by steveohara