steveohara/j2mod #14: WriteMultipleRegisters (Function Code : 0x10) does not work.

nices96, 2016-04-27:
I created a mockup modbus slave(TCP Server) using _**ModbusTCPListener**_ class and test WriteMultipleRegisters function, but error occurred with "Illegal Data Address" message.

I think it is problem with **WriteMultipleRegistersRequest.readData()**'s _**m_Reference = input.readShort();**_.

I changed to **_input.readUnsignedShort()_** and checked it works fine.

steveohara, 2016-04-29:
I added a test case for this and it works OK - are you using a register number over 32767?

steveohara, 2016-04-29:
Added fixes for this and other places where the library should be using unsigned shorts
The latest snapshot has the fixes - could you let me know if you see any issues
https://oss.sonatype.org/content/repositories/snapshots/com/ghgande/j2mod/2.0-SNAPSHOT/j2mod-2.0-20160429.141401-1.jar

closed 2016-04-29 by steveohara

nices96, 2016-04-29:
Yes, I did use over 40000 as a register number. Thanks.