steveohara/j2mod #15: Support as OSGi Bundle

amitjoy, 2016-05-05:
It would be helpful if you can provide this library as an OSGi bundle so that it can be made available in lots of OSGi based frameworks.

steveohara, 2016-05-05:
Which j2mod package(s) did you have in mind to expose in the bundle?

amitjoy, 2016-05-05:
I believe it would be better to expose all the necessary packages which contain classes needed to write Modbus client code.

steveohara, 2016-05-06:
Amit, that isn't practical or actually that helpful for OSGi in my opinion.
I think exposing the Façade package makes sense as it aggregates a lot of boilerplate but creating interfaces for all the constituents just adds cruft to the library.

amitjoy, 2016-05-06:
You are right that facade package can be the most preferred one to be exposed but as j2mod is an advanced version of jamod, people still can use the old classes to do the same stuff. I would suggest you to expose all the relevant packages except the utilities if present.

amitjoy, 2016-05-16:
I am just wondering if there is any plan of providing this library as an OSGi bundle?

steveohara, 2016-05-16:
Hi Amit, I'm afraid I'm fully committed to other things at the moment but I would welcome any patches so please feel free to put a pull request together. Otherwise, it will have to wait a little while until I can back to j2mod

julie-ab, 2016-05-16:
@amitjoy - I've been fully occupied with my day job, but if you can explain what's involved I may have more free time than Steve.

amitjoy, 2016-05-16:
@steveohara I understand. I am also a bit busy recently. I would try to convert j2mod as an OSGi bundle so that people using OSGi can make use of this library as an OSGi bundle.

@julie-ab My idea is to incorporate j2mod in Java- and OSGi-based IoT Frameworks and that's why, it would be good to make it available as a standalone OSGi bundle. If j2mod needs any dependency, it would contain that dependency within the bundle jar as its bundle classpath and in the same time, the bundle jar would expose its public packages so that people can make use of only the j2mod classes. So, the bundle can be used as a simple library jar as well as an OSGi bundle.

Let me know if you are interested into my idea.

cnoelle, 2018-06-18:
Any plans to merge this? Would be great to have OSGi support in j2mod.

By the way, follwing up on the discussion above, exporting all packages seems to be the only sensible thing to do here, since applications may need to access classes from all packages. The commit does exactly that.

steveohara, 2018-07-19:
Code merged into snapshot

closed 2018-07-19 by steveohara