steveohara/j2mod #16: Serial/RTU slower on 2.x

axuan25, 2016-05-07:
Hello,

For starters, I would like to thank all of the people who are working on improving j2mod, notable in the exception handling and logging aspects. The 2.x version looks much cleaner and it is a relief to move away from RxtxComm !

That being said, I just migrated to j2mod 2.0, and using the exact same code for my RTU tests, I noticed the time tripled. A single connection to read 2 registers takes ~400ms on j2mod 1.06, but takes 2s 800ms using j2mod 2.x.

Has anyone else noticed this behavior upon moving 2.0?

steveohara, 2016-05-10:
I've been following through when the timeouts are applied to the comms port and noticed something that might be contributing to this.
The ModbusSerialTransport readResponse and writeResponse methods apply the timeouts before calling the actual stream read on the JSerialComms port. Looking at the source code for jSerialComms I can see that this injects a 200ms delay because it detects that te port is open (I don't know why it does this).
So for a single write/read transaction, this would effectively inject a 400ms delay.
I can't account for the 2 seconds you are seeing though.
I'll do some testing and see if I can re-work the way the timeouts are applied to get better performance.

steveohara, 2016-05-10:
I've moved the setting of timeouts and tested it against the RTU kit I have here and the perfomance is about 8x improved.
This snapshot has the changes in it https://oss.sonatype.org/content/repositories/snapshots/com/ghgande/j2mod/2.0-SNAPSHOT/j2mod-2.0-20160510.212924-1.jar
Can you try it out and let me know that it's functioning OK for you

closed 2016-05-10 by steveohara

steveohara, 2016-05-19:
Found another source of slowness in jSerialComm
Every time you open a connection, there is a 500ms delay