steveohara/j2mod #27: Amend TCPMasterConnection socket handling

djbr-david, 2016-07-15:
When attempting a connection in the above class the Socket creation isn't respecting the timeout period. The result is that if a connection is attempted to an unavailable host the code hangs for 30-40s before issuing a timeout exception. The connect method should probably use:
```
        if (!isConnected()) {
            logger.debug("connect()");

            socket = new Socket();
            socket.setReuseAddress(true);
            socket.setSoLinger(true, 1);
            socket.setKeepAlive(true);
            setTimeout(timeout);

            socket.connect(new InetSocketAddress(address, port), timeout);

            prepareTransport(useRtuOverTcp);

            connected = true;
        }
```
which does respect the timeout.

steveohara, 2016-07-15:
I've made the suggested changes and added a test case.
Can you try out https://oss.sonatype.org/content/repositories/snapshots/com/ghgande/j2mod/2.2.0-SNAPSHOT/j2mod-2.2.0-20160715.114835-1-javadoc.jar

djbr-david, 2016-07-15:
Thats working for me. Thanks Steve.
BTW: The above link is for the javadoc content. The code content is: [https://oss.sonatype.org/content/repositories/snapshots/com/ghgande/j2mod/2.2.0-SNAPSHOT/j2mod-2.2.0-20160715.114835-1.jar]

steveohara, 2016-07-15:
Whoops...

closed 2016-07-15 by steveohara