steveohara/j2mod #40: Serial port connection

Kosmachevskiy, 2016-12-08:
_Library version: 2.3.1_

Hi!

I have some trouble with ModbusSerialMaster. I got a NPE during reading registers.

 
My code:

`

        SerialParameters parameters = new SerialParameters();

        parameters.setPortName("/dev/ttyUSB1");
        parameters.setBaudRate(9600);
        parameters.setDatabits(8);
        parameters.setStopbits(1);
        parameters.setParity("none");
        parameters.setEncoding("rtu");
        parameters.setEcho(false);

        ModbusSerialMaster master = new ModbusSerialMaster(parameters);

        master.connect();
        master.readMultipleRegisters(1, 0, 100);
        master.disconnect();
`
And stack:

`Exception in thread "main" java.lang.NullPointerException
	at com.ghgande.j2mod.modbus.net.SerialConnection.setComPortTimeouts(SerialConnection.java:221)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransport.setTimeout(ModbusSerialTransport.java:142)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransport.open(ModbusSerialTransport.java:131)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransport.writeMessage(ModbusSerialTransport.java:73)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransaction.execute(ModbusSerialTransaction.java:143)
	at com.ghgande.j2mod.modbus.facade.AbstractModbusMaster.readMultipleRegisters(AbstractModbusMaster.java:236)`

steveohara, 2016-12-29:
It shouldn't be throwing an NPE but I suspect it's because it cannot create a JSerialComm wrapper around "/dev/ttyUSB1"
Does `SerialPort.getCommPort("/dev/ttyUSB1")` return anything?

notsyncing, 2017-01-12:
I ran into the same(or similar) problem recently, and finally found the cause is current user didn't have the permission to access the serial port (not in ```dialout``` group).
By the way, would you mind add some checks when opening the serial port? Making it fails early at opening the serial port instead of fails at first operation could reduce the difficulty of debugging such problems.

Kosmachevskiy, 2017-01-12:
Problem really with permissions. Thanks.

closed 2017-01-12 by Kosmachevskiy

steveohara, 2017-01-12:
I've created https://github.com/steveohara/j2mod/issues/45 so that I don't forget about it