steveohara/j2mod #50: NullPointerException when trying to communicate between Windows and Linux

Ahmid, 2017-04-19:
Hello,

Simply, I am running Modbuspal.java on a windows machine, where I have created a slave and filled some coils and registers.

Then from the same windows machine, I used j2mod in order to read coils/registers, the operation wen successfully. 

The problem is that when I ran the same code on a linux (ubuntu) virtual machine, a null pointer exception is being released everytime, the connection phase is succeeding, but the reading is not.

The code is:

import java.net.InetAddress;
import java.util.logging.Level;
import java.util.logging.Logger;

import com.ghgande.j2mod.modbus.Modbus;
import com.ghgande.j2mod.modbus.ModbusException;
import com.ghgande.j2mod.modbus.io.ModbusTCPTransaction;
import com.ghgande.j2mod.modbus.msg.ReadCoilsRequest;
import com.ghgande.j2mod.modbus.msg.ReadCoilsResponse;
import com.ghgande.j2mod.modbus.msg.ReadInputDiscretesRequest;
import com.ghgande.j2mod.modbus.msg.ReadInputDiscretesResponse;
import com.ghgande.j2mod.modbus.msg.ReadMultipleRegistersRequest;
import com.ghgande.j2mod.modbus.msg.ReadMultipleRegistersResponse;
import com.ghgande.j2mod.modbus.net.TCPMasterConnection;

//Process
public class ModbusTest {

	/* The important instances of the classes mentioned before */
    protected static TCPMasterConnection con = null; //the conection
    /* Variables for storing the parameters */
    protected static InetAddress addr = null; //the slave's address
    protected static final int port = 502;
    protected static final int ref = Integer.parseInt("1", 16); //the reference; offset where to start reading from
    protected static final int count = 100; //the number of DI's to read
    protected static final int repeat = 1; //a loop for repeating the transaction
    static int SlaveAddr=1;
	
	
	public static void main(String[] args) {
		
		try {
            addr = InetAddress.getByName("130.190.97.101");
            //Coils
            ReadCoilsRequest RCreq = new ReadCoilsRequest (3,1);
            ReadCoilsResponse RCres = new ReadCoilsResponse();
            
            //Registers
            ReadMultipleRegistersRequest RRreq = new ReadMultipleRegistersRequest (4,1);
            ReadMultipleRegistersResponse RRres = new ReadMultipleRegistersResponse();
            
            RCreq.setUnitID(SlaveAddr); //set Slave Address  
            RCres.setUnitID(SlaveAddr); //set Slave Address
            
            RRreq.setUnitID(SlaveAddr);
            RRres.setUnitID(SlaveAddr);
            
            
            //2. Open the connection
            con = new TCPMasterConnection(addr);
            con.setPort(5555);
            con.connect();
            
            ModbusTCPTransaction trans = null;
            trans = new ModbusTCPTransaction(con);
            
            
            trans.setRequest(RCreq);
            trans.execute();
            RCres = (ReadCoilsResponse) trans.getResponse();
            System.out.println("Connected to = "+ addr + " " + con.isConnected() + " /Start Register " + Integer.toHexString(ref));
            System.out.println("The value READ (coils): " + RCres.getCoils().toString());
            
            trans.setRequest(RRreq);
            trans.execute();
            RRres = (ReadMultipleRegistersResponse) trans.getResponse();
            for (int n = 0; n < RRres.getWordCount(); n++) {
                System.out.println("Word = " + RRres.getRegisterValue(n));
            }
            
            
            //6. Close the connection
            con.close();
        } catch (Exception ex) {
            Logger.getLogger(ModbusTest.class.getName()).log(Level.SEVERE, null, ex);
        }
    }

}

steveohara, 2017-06-07:
It's not clear from your code if the exception is because RCres is null or the result of RCres.getCoils(). Could you tell me which it is?
Also, can you turn on debug logging to see why an exception is not being thrown for an empty response.

steveohara, 2017-09-10:
It has been 2 months and nobody else has reported an issue so I'm closing this ticket

closed 2017-09-10 by steveohara