steveohara/j2mod #52: WARNUNG: Error com.ghgande.j2mod.modbus.ModbusIOException:I/O exception - failed to read

WolfgangFahl, 2017-07-19:
Thank you for keeping this library with an excellent api alive.
I am having issues comparable to:
- https://sourceforge.net/p/j2mod/discussion/general/thread/726f3239/
- https://stackoverflow.com/questions/26621299/java-modbus-rtu-master-example-code

with the following code on a raspberry PI using /dev/ttyUSB0 with a UART USB-connector applied. The connection works with QModBus at 9600 baud, 8 databits 1 stop bit and no parity.
What might be wrong in this setup and how can I debug it?

[image]

```
import com.ghgande.j2mod.modbus.util.SerialParameters;

/**
 * SimpleEvSE Modbus
 * 
 * @author wf
 *         // https://github.com/steveohara/j2mod
 */
public class SModbus {

  String portName;
  private SerialParameters serialParams;
  private ModbusSerialMaster master;
  private int timeOut;

  /**
   * construct me for the given port
   * 
   * @param portName
   * @param timeOut
   */
  public SModbus(String portName, int timeOut) {
    this.portName = portName;
    this.timeOut = timeOut;
    // http://jamod.sourceforge.net/kb/serial_master_howto.html
    // https://stackoverflow.com/questions/26621299/java-modbus-rtu-master-example-code
    serialParams = new SerialParameters();
    serialParams.setPortName(portName);
    serialParams.setBaudRate(9600);
    serialParams.setDatabits(8);
    serialParams.setParity("None");
    serialParams.setStopbits(1);
    serialParams.setEncoding("ascii");
    serialParams.setEcho(false);
    master = new ModbusSerialMaster(serialParams, this.timeOut);
  }

  /**
   * dump registers
   * 
   * @throws Exception
   */
  public void dumpRegisters() throws Exception {
    master.connect();
    // https://github.com/steveohara/j2mod/pull/48
    AbstractSerialConnection con = master.getConnection();
    System.out.println(String.format(
        "connected to %s %s at %5d baud %2d data bits %2d stop bits\n available ports:%s",
        con.getDescriptivePortName(), portName, con.getBaudRate(),
        con.getNumDataBits(), con.getNumStopBits(),
        con.getCommPorts().toString()));
    master.setRetries(0);
    int ref = 1000;
    int count = 2;
    Register[] regs = master.readMultipleRegisters(ref, count);
    int i = ref;
    for (Register reg : regs) {
      System.out.println(String.format("%3d: %3d", i++, reg.getValue()));
    }
    master.disconnect();
  }
}
```

WolfgangFahl, 2017-07-19:
just found https://sourceforge.net/p/j2mod/discussion/general/thread/0afca16d/ and trying the debug and Encoding options

```
 serialParams.setEncoding(Modbus.SERIAL_ENCODING_RTU);
    serialParams.setEcho(true);
    System.setProperty("com.ghgande.modbus.debug", "true");
```

Play3r21, 2017-07-27:
I believe part of the problem is that you aren't setting up your SerialParameters object correctly.

Here's a class that I used to test this library on a VirtualBox VM running Ubuntu 14.04LTS with a USB-to-RS232 converter (ttyUSB0)

import com.ghgande.j2mod.modbus.*;
import com.ghgande.j2mod.modbus.facade.ModbusSerialMaster;
import com.ghgande.j2mod.modbus.net.*;
import com.ghgande.j2mod.modbus.procimg.Register;
import com.ghgande.j2mod.modbus.util.*;

public class ModbusMaster {
	
	/**
	 * @param args
	 */
	public static void main(String[] args) {
		/* The important instances of the classes mentioned before */
		ModbusSerialMaster serialMaster = null; // the connection
		
		/* Variables for storying the parameters */
		String portname = "/dev/ttyUSB0"; // the name of the serial port to be used
		int unitID = 2; // the unit identifier we will be talking to
		int startingRegister = 0; // the reference, where to start reading from
		int registerCount = 4; // the count of the input registers to read
		Register[] slaveResponse = new Register[registerCount];

		try {
			/* Setup the serial parameters */
			SerialParameters parameters = new SerialParameters();
			parameters.setPortName(portname);
			parameters.setBaudRate(9600);
			parameters.setDatabits(8);
			parameters.setParity(AbstractSerialConnection.NO_PARITY);
			parameters.setStopbits(AbstractSerialConnection.ONE_STOP_BIT);
			parameters.setEncoding(Modbus.SERIAL_ENCODING_RTU);
			parameters.setEcho(false);

			/* Open the connection */
			serialMaster = new ModbusSerialMaster(parameters);
			serialMaster.connect();

		} catch (Exception exception) {
			exception.printStackTrace();
		}

		/* Read the first four holding registers */
		try {
			slaveResponse = serialMaster.readMultipleRegisters(unitID, startingRegister, registerCount);
			for (int i = 0; i < slaveResponse.length; i++) {
				System.out.println("reg" + i + " = " + slaveResponse[i]);
			}
		} catch (ModbusException e) {
			e.printStackTrace();
		}
		/* Close the connection */
		serialMaster.disconnect();

	} // main

} // class ModbusMaster

Please note that this is very crude code and I would not use this in a production setting, but I was using this as a simple test of the serial capabilities of this library on Linux.

Something else you may be running up against is privilege restrictions on the port. Check the [jSerialComms page](http://fazecast.github.io/jSerialComm/) for instructions on placing your user into the correct group (e.g. dialout, tty, etc.)

I hope this helps!

WolfgangFahl, 2017-07-28:
thankx. These modifications seem to make the difference:

```
@@ -35,10 +35,10 @@ public class SModbus {
     serialParams.setPortName(portName);
     serialParams.setBaudRate(9600);
     serialParams.setDatabits(8);
-    serialParams.setParity("None");
-    serialParams.setStopbits(1);
+    serialParams.setParity(AbstractSerialConnection.NO_PARITY);
+    serialParams.setStopbits(AbstractSerialConnection.ONE_STOP_BIT);
     serialParams.setEncoding(Modbus.SERIAL_ENCODING_RTU);
-    serialParams.setEcho(true);
+    serialParams.setEcho(false);
```

BTW code comments can be created with three backticks or cliicking the "insert code" icon 
[image]

closed 2017-09-10 by steveohara

adrian1388, 2019-02-21:
I was facing this issue...
I cloned the code and installed via `mvn install -DskipTests` and magically worked.

anilStartele, 2020-05-18:
I am also facing the same issue.. Will somebody help me?