steveohara/j2mod #54: Getting j2mod/jSerialComm to work on Android

Play3r21, 2017-07-31:
Has anyone been able to successfully get j2mod and its dependency jSerialComm to work on Android? I'm facing the same issue that the users in [jSerialComm's issues thread 45](https://github.com/Fazecast/jSerialComm/issues/45) were facing. I've been searching for an answer for more than a week and have gotten nowhere.

I've tried the following (plus many others, all of which did not help):
- Google's [Add Build Dependencies](https://developer.android.com/studio/build/dependencies.html), which consisted of me placing the JARs for j2mod, jSerialComm, and slf4j into my project's app/libs/ directory. I also tried removing the jSerialComm JAR and adding it's Gradle dependency directive for Maven, _compile 'com.fazecast:jSerialComm:1.3.11'_.
- The marked solution in this Stackoverflow post [on how to include .so files](https://stackoverflow.com/questions/27421134/system-loadlibrary-couldnt-find-native-library-in-my-case) (which I tried to do instead of using the jSerialComm JAR file)
- The marked solution in this Stackoverflow post [on getting jSerialComms to work in Android](https://stackoverflow.com/questions/35213730/android-dlopen-failed-file-offset-for-the-library/35214450#35214450)

I'm working with a device running Android 5.1.1 (API 22) on an ARM Cortex-A9 SoC with 5 physical serial ports, labelled ttyAMA0 through ttyAMA4 in /dev/. SELinux is set to _Disabled_.

I know that this question is better suited for the jSerialComm issues threads, but it's already been asked there and no one has provided a viable solution.

Thank you very much for the help.

steveohara, 2017-09-10:
You are correct, this question isn't suitable for this forum, howerver...

The serial comms is abstracted in j2mod (`AbstractSerialConnection`) so if you can't find a workaround, you could always implement your own serial connection using an Android compatible library.
Take a look at the `ModbusSerialMaster` to see how to use your own comms connection.

closed 2017-09-10 by steveohara

orensharon, 2018-02-16:
This is the correct answer to make it work on android device:
https://github.com/Fazecast/jSerialComm/issues/36

* Clone jSerialComm to your pc. 
* Edit the code as @adryx92 said in the above link.
* Inject this jar to j2mod lib and create new  jar of j2mod.
* Add j2mod jar to Android project.

****
 To inject jSerialComm to j2mod remove jSerialComm depenedency from j2mod's pom.xml file and import  jSerialComm.jar from the resource folder

Make sure maven doesn't import the source from .m2 library. I had to delete jSerialComm from .m2 folder to make it work.
You also need to copy all the files from jSerialComm.jar/Android to jniLibs folder in your androids project.

For build jSerialComm lib I used eclipse (it didn't work with intelij) with:
* Java 1.6 (didn't work with Java 1.8)
* Gradle 3.5 

One more thing is to give root access to the serial port in the Android device (you can take a look at this :https://github.com/chrisdew/android-serialport-api/blob/10cb9c4d6dd7e4357473d1438530dd5639eb5e12/android-sercd/project/src/gnu/sercd/Sercd.java#L69)