steveohara/j2mod #62: How to only listen to one specific UnitID with serial RTU slave?

NogginBops, 2017-09-29:
I'm trying to write an application that is supposed to emulate a slave on a specific address. So far I have gotten the program to receive data through the serial port and it sends back som messages to the master. But I sends a lot of exception messages because it receives messages ment for another unit. This is not good for the master program as it only expects one answer but gets a response from two units.

Is there a way to stop this behavior?
To create the slave I do: `ModbusSlave slave = ModbusSlaveFactory.createSerialSlave(params);`
I create a ProcessImageImplementation like this: ``
And add it to the slave like this: ``

Is there something that I am doing wrong, something about the library I am using in a wrong way?

Otherwise this is a very good feature to have as it allows for simulating one of many devices on  a network.

steveohara, 2017-10-24:
I think logically there must be something very wrong here. 
A Modbus Slave should never react to requests that are not for it's Unit ID.
Can you paste your code into this ticket so I can see what you're doing?

NogginBops, 2018-04-18:
This is what I do to create a slave.

```java
slave = ModbusSlaveFactory.createSerialSlave(params);
img = factory.createProcessImageImplementation();
createRegisters(img); // Allocates registers
slave.addProcessImage(slaveID, img);
slave.open();
```

steveohara, 2018-04-19:
The filtering of messages is done in the listener. Take a look at line 166-167 of AbstractModbusListener.
This infers that you must set a UnitID for your ProcessImage. Calling factory.createProcessImageImplementation() creates a SimpleProcessImage with a unit ID of 0.

A better way to do this would be to do the following;
```
slave = ModbusSlaveFactory.createSerialSlave(params);
img = new SimpleProcessImage(slaveID);
createRegisters(img); // Allocates registers
slave.addProcessImage(slaveID, img);
slave.open();
```

I think the j2mod code in this area needs simplification i.e. why does a process image need to know it's Unit ID if when it is added to the Slave it is allocated to s Slave ID then?
There is a refactor on the way that will remove the old ModbusCoupler and tidy up the handling off unit IDs for slaves but in the meantime, the above should work for you.

NogginBops, 2018-04-25:
Never mind, this issue is not fixed...

And looking at the filtering code I'm a bit confused as it clearly does not filter any messages.
```java
if (spi == null || (spi.getUnitID() != 0 && request.getUnitID() != spi.getUnitID())) {
    response = request.createExceptionResponse(Modbus.ILLEGAL_ADDRESS_EXCEPTION);
}
```

It clearly always sends an exception when there is no matching process image?
Or is there something I am missing again?

steveohara, 2018-04-29:
Can you try the latest snapshot and let me know how it goes please

NogginBops, 2018-04-29:
I did a quick tests now to see if it worked by just plugging in the snapshot jar and it did not seem to work as correctly. As the master was now consistently getting read timeouts.
I was not able to check if this was a configuration issue. But as this communication has worked with the fix I made so I doubt that is the problem.
Sorry for the limited info, I can do a more thorough debugging of this on Wednesday.

NogginBops, 2018-05-03:
Ok so it seems to work somewhat. But now I'm experiencing another problem where reading or writing sometimes takes more than 1 second. And it seems really weird as I have set a read timeout of 300ms, I am running linux but from what I have gathered the read timeout should still work?

steveohara, 2018-05-03:
There is a delay you can set on the `ModbusSerialTrsansaction` object using `setTransDelayMS` - the default value is 0, it used to be 50msec.

You can't access this via the `ModbusSerialMaster`, you would have to construct requests using the mechanisms shown in the test `ReadInputRegistersTest`

That's not ideal so perhaps an overloaded version of the constructor to `ModbusSerialMaster` where you can specify this value?

NogginBops, 2018-05-03:
I think that this should be standard for all RTU communication as it is specified in the specification. The time is computable from the baudrate.

closed 2018-05-10 by steveohara