steveohara/j2mod #63: Got java.util.IllegalFormatConversionException in exceptions throws

marlontrujillo1080, 2017-10-04:
Stacktrace:

```
java.util.IllegalFormatConversionException: d != com.ghgande.j2mod.modbus.ModbusIOException
	at java.util.Formatter$FormatSpecifier.failConversion(Formatter.java:4302)
	at java.util.Formatter$FormatSpecifier.printInteger(Formatter.java:2793)
	at java.util.Formatter$FormatSpecifier.print(Formatter.java:2747)
	at java.util.Formatter.format(Formatter.java:2520)
	at java.util.Formatter.format(Formatter.java:2455)
	at java.lang.String.format(String.java:2940)
	at com.ghgande.j2mod.modbus.ModbusException.<init>(ModbusException.java:59)
	at com.ghgande.j2mod.modbus.ModbusIOException.<init>(ModbusIOException.java:60)
	at com.ghgande.j2mod.modbus.io.ModbusTCPTransaction.execute(ModbusTCPTransaction.java:176)
```

jeffgrossmann144, 2017-10-18:
Root cause is a bad ModbusIOException being created in ModbusTransaction.execute at line 176.   Replacing with the following line of code fixes the issues.

`                    throw new ModbusIOException(String.format("Executing transaction failed (tried %d times)", retryLimit), ex);
`

steveohara, 2017-10-24:
This issue was already fixed in the 2.3.7 version (see the release notes)

closed 2017-10-24 by steveohara

vermicious, 2017-10-31:
Fair enough, but 2.3.7 is still not available in the central maven repository.

steveohara, 2017-11-01:
Sorry, it's in the 2.3.7-snapshot release.
Nearly all changes go into the snapshot and stay there for a while until I have enough confidence that they should go into a production release.
I will release 2.3.7 today.