steveohara/j2mod #64: Modbus TCP Slave / How to detect process image change

dannaughton, 2017-11-07:
(I posted this on the sourceforge site, but I saw you moved here - sorry for duplicate post)

I have question about modbus TCP slave. I have the simple tutorial working (the one from the Wiki) and masters can connect to it and read/write the process image

Question:
On my slave device, what's the best way to detect if the process image has been changed (by the master of course).

steveohara, 2017-11-08:
The best way is to implement you own ProcessImage and handle the updates. You could do that explicitly or extend SimpleProcessImage.
You then need to add the process image to the slave you have started.

dannaughton, 2017-11-09:
Thanks Steve,

That was helpful.  Implementing my own ProcessImage is beyond my skill level.  I looked into extending SimpleProcessImage and using Observable to handle changes, but that creates this Multiple Inheritance requirement that is also beyond my skill level to work through.

Everything seems to be working well, and stable.  I'll keep looking to find someway to detect and handle when the ProcessImage is changed.  There is an Observer interface that's included in j2mod.  Not sure what the differences are between that and the normal Observer - but in either case, I'm not sure where to plug it in.

I'll report back - Thanks for your help

dannaughton, 2017-11-09:
I think I have a handle on it.  It looks like using the ObservableDigitalOut and ObservableRegister is the way to go, rather than trying to extend SimpleProcessImage.

The following seems to work well (so I'm good - thanks for your help):

       spi = new SimpleProcessImage();
       ObservableDigitalOut odo = new ObservableDigitalOut();
       odo.set(true);
       odo.addObserver(this);
       spi.addDigitalOut(odo);

	@Override
	public void update(Observable o, Object arg) {
		System.out.println("Got notification from odo");
		
	}

steveohara, 2017-11-09:
I learn something new every day! I've never used the Observable features of j2mod before.
Thanks

closed 2017-11-09 by steveohara

pan-henryk, 2018-06-02:
It seems not to work in 2.4.1 and 2.5.0 with ObservableRegisters.
I am doing:
`                    SimpleProcessImage spi = new SimpleProcessImage(unitId);
                    ObservableRegister or = new ObservableRegister();
                    or.setValue(bb.getShort());
                    or.addObserver(this);
                    spi.setInputRegister(address,or);`
I can properly read values for that register (so the setup is ok), but if I write to register from master, update method of observer is never called.
At the message level communication seems to work fine (slave receives write request and responds to it without any errors). I am using Modbus TCP for that.
Is there a chance that refactoring slave broke it?