steveohara/j2mod #69: BIN encoding in serial connection

dyosick, 2018-03-20:
Is the BIN encoding supported for the serial connections?
From [this](https://github.com/steveohara/j2mod/blob/development/src/main/java/com/ghgande/j2mod/modbus/util/SerialParameters.java#L428) I concluded, _j2mod_ doesn't support. But _jamod_ [docs ](Implementation)say that it does (see section _Implementation_, first _Note_).

steveohara, 2018-03-20:
No this was a casualty of the huge refactoring that brought about v2.

My main objective is to make sure the library is standards compliant, robust and performant.
The BIN protocol was created by the original author of jamod and is not known outside that realm, so it won't be coming into j2mod I'm afraid.

dyosick, 2018-03-20:
Thanks for the quick response.

closed 2018-03-20 by dyosick