steveohara/j2mod #70: Multiple reads of the same function code

orensharon, 2018-04-10:
The registers on my device as following:
* Holding registers: ref: 0, length 18
* Holding registers: ref: 100, length 18
* Holding registers: ref: 200, length 1

I can't read it in 1 read request because the protocol limit is 125 registers per request.
When I'm trying to read on by other, like this:
registers = master.readMultipleRegisters(1, 0, 18);
registers = master.readMultipleRegisters(1,100, 0);

Something is not correct, I got delays between the reads.

steveohara, 2018-04-10:
There isn't anything wrong with what you're doing except the obvious issue that you have specified a register count of zero in the second example.

What type of transaction are you using? (serial, TCP, RTU over TCP)

When you say "there are delays between read" I don't understand what you mean. Are you saying that each subsequent request after the first is delayed before sending?

If you are using a serial transaction, are you setting a transaction delay with setTransDelayMS

orensharon, 2018-04-10:
Sorry, I fixed the issue to the correct length: 
`registers = master.readMultipleRegisters(1,100, 1);`

I'm using Serial RTU.
setTransDelayMS seems to be a good start. How do I set it?
Here is monitor output i use to see the problem:
[image]

Thanks!

orensharon, 2018-04-10:
Update:
I changed (hard coded) TransDelayMS and now it working good!

closed 2018-04-10 by steveohara