steveohara/j2mod #76: Serial slave not reading response from other slave

NogginBops, 2018-05-14:
## Expected Behavior

The slave should detect that the master request is not destined to the slave. After this the slave should expect there to come a response on the serial line and read it. Otherwise should timeout and expect another request.

## Actual Behavior

The slave detects that the message it not for it, and correctly ignores it. But the slave does not expect the response and thinks that this is supposed to be another request.

## Steps to Reproduce the Problem

  1. Connect one master to two slaves on a serial bus.
  2. Send read requests to one of the slaves.
  3. Look at debug printouts on the other master.

## Specifications

  - Version: 2.5.1-SNAPSHOT

NogginBops, 2018-05-15:
This looks like it is a little bit more complicated as the master could be resending it's request. If the slave is expecting a response there will be some error reading the response back. What really should be done it early filtering of messages based on their ID, and then just reading all data until there is a 3.5T gap. That way there is no need to keep track of the state of the buss. Just respond to the messages meant for the slave and ignore the rest.

steveohara, 2019-01-07:
Merged changes into snapshot

closed 2019-01-07 by steveohara

NogginBops, 2019-01-09:
Oh! Sorry I never fixed my pull-request! Good that you fixed the problems with it 👍