steveohara/j2mod #82: Is ObservableRegister broken?

pan-henryk, 2018-06-02:
I am trying to detect changes in my slave image in a way similar to described here:
https://github.com/steveohara/j2mod/issues/64

My code looks like this:

`                   
                   SimpleProcessImage spi = new SimpleProcessImage(unitId);
                   ObservableRegister or = new ObservableRegister();
                    or.setValue(bb.getShort());
                    or.addObserver(this);
                    spi.setInputRegister(address,or);`

I can read this register from master (so general setup is ok), but if I try to write to it, update method of observer is never called.
Is there a chance that slave refactoring broke it?

I am using Modbus TCP.

steveohara, 2018-06-05:
Could well be.
Unfortunately we don't use this pattern at 4NG so it doesn't get a lot of love from us. If you could put together a unit test, I'll take a look at figuring out the problem.

pan-henryk, 2018-06-08:
I have hacked simple observable register test:
https://github.com/steveohara/j2mod/pull/84

steveohara, 2018-07-19:
Test case added

closed 2018-07-19 by steveohara