steveohara/j2mod #90: Is there any plans to update jSerialComm to the latest version (2.3.0)?

dvk2018, 2018-12-10:


dvk2018, 2018-12-10:
Also, is it a convenient way for this project to update the jSerialComm version every time j2mod has been released?

steveohara, 2018-12-10:
There is now!
I always want to keep closely aligned with JSerialComm so I will update and test j2mod with 2.3.0 next week when I'm back in the UK

steveohara, 2018-12-16:
I've updated to V2.3.0 - there is a possibility that there is an issue with this version of the comms library because I had to alter the blocking strategy on reads to make it work.
If you could try this snapshot version out I would appreciate some feedback
https://oss.sonatype.org/content/repositories/snapshots/com/ghgande/j2mod/2.5.3-SNAPSHOT/j2mod-2.5.3-20181216.230213-3-sources.jar

closed 2018-12-16 by steveohara

steveohara, 2018-12-17:
Many thanks

dvk2018, 2018-12-19:
Hi @steveohara,
When are you going to release 2.53 version?

steveohara, 2018-12-19:
Just been released - if you wait an hour or so, it will appear on Maven Central