steveohara/j2mod #96: ClassCastException when timeout??

niclash, 2019-02-22:
## Expected Behavior
Correct decoding of message

## Actual Behavior
java.lang.ClassCastException: com.ghgande.j2mod.modbus.msg.WriteSingleRegisterResponse cannot be cast to com.ghgande.j2mod.modbus.msg.ReadInputRegistersResponse
	at com.ghgande.j2mod.modbus.facade.AbstractModbusMaster.readInputRegisters(AbstractModbusMaster.java:209)

## Steps to Reproduce the Problem

I am not exactly sure when it happens. It is very intermittent, but it seems to be a multi-threading issue and that the "read" will accidentally be "lost" in one thread (no answer) and the "write" thread gets the read result and gets confused...

I am running j2mod RTU mode. I am pretty sure I have seen somewhere that j2mod is thread-safe, and for the most part my multi-threaded app is doing fine (I don't depend on the resolution of this bug)

## Specifications
  - Version: 2.5.1
  - Platform: Linux
  - Subsystem: 
       $ uname -a
       Linux devdesk 4.19.0-1-amd64 #1 SMP Debian 4.19.12-1 (2018-12-22) x86_64 GNU/Linux

bthulu, 2019-11-10:
me, too, TCP mode.
java.lang.ClassCastException: com.ghgande.j2mod.modbus.msg.WriteSingleRegisterResponse cannot be cast to com.ghgande.j2mod.modbus.msg.ReadMultipleRegistersResponse
	at com.ghgande.j2mod.modbus.facade.AbstractModbusMaster.readMultipleRegisters(AbstractModbusMaster.java:238) ~[j2mod-2.5.5.jar!/:2.5.5]
	at com.ghgande.j2mod.modbus.facade.AbstractModbusMaster.readMultipleRegisters(AbstractModbusMaster.java:426) ~[j2mod-2.5.5.jar!/:2.5.5]
	at bthulu.commons.combine.modbus.MbusMaster.readBytes(MbusMaster.java:233) ~[jcoutil-0.0.5.jar!/:na]
	at bthulu.commons.combine.modbus.MbusMaster.readShort(MbusMaster.java:202) ~[jcoutil-0.0.5.jar!/:na]
	at bthulu.commons.combine.modbus.MbusMaster.readShort(MbusMaster.java:198) ~[jcoutil-0.0.5.jar!/:na]
	at domiyi.paper.obp.prepare.PreparePlcManager.readRemoteControl(PreparePlcManager.java:107) ~[classes!/:na]
	at domiyi.paper.obp.prepare.PaperOnService.lambda$saveOnPrepareDisable$3(PaperOnService.java:160) ~[classes!/:na]
	at org.springframework.scheduling.support.DelegatingErrorHandlingRunnable.run(DelegatingErrorHandlingRunnable.java:54) ~[spring-context-5.1.9.RELEASE.jar!/:5.1.9.RELEASE]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [na:1.8.0_231]
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308) [na:1.8.0_231]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180) [na:1.8.0_231]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294) [na:1.8.0_231]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [na:1.8.0_231]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [na:1.8.0_231]
	at java.lang.Thread.run(Thread.java:748) [na:1.8.0_231]

steveohara, 2019-11-16:
I'm wondering if you are multiplexing calls to `ModbusTCPMaster` or `ModbusSerialMaster` because those are not thread safe. You should either synchronise your calls to these or not share them across threads.

steveohara, 2019-11-16:
Fixed in 2.5.8

closed 2019-11-16 by steveohara