steveohara/j2mod #101: ReadRequest

YaoChiXiaoMi, 2019-09-02:
I found this problem after using j2mod for a while.The program that collects modbus serial data runs for a while.

Occasionally such a phenomenon（It can happen in a few months）：
The hexadecimal data sent and the hexadecimal data received are in compliance with the modbus protocol.
However, the read data method of j2mod.jar, such as ReadInputRegistersRequest/ReadMultipleRegistersRequest, will report an exception.
So that data cannot be collected normally,At this time, you need to re-initialize the ModbusTransaction to reacquire the data.

So, have you discovered this problem before? And has the latest j2mod version been fixed?
Can you tell me what to do to avoid this problem?
system：Windows2012
Serial port：Virtual serial port
j2mod version：2.5.4

steveohara, 2019-09-02:
I can't say that I've seen this issue reported.
To be honest, good practise would be to capture the exception and then re-establish the connection.
Can you at least send me the exception you are seeing?

YaoChiXiaoMi, 2019-09-02:
Ok, I will send you the relevant information about the error after the next time this happens again.

YaoChiXiaoMi, 2019-09-09:
Last night, I had the problem I told you last time.
Because I am a virtual serial port on the cloud server, the data frame is transmitted through the tcp connection.
and then last night, the tcpserver on the cloud server is abnormally connected, and then receives abnormal data frames continuously.
Just like the picture below

Then the j2mod java package reported a CRC check error, which is understandable, after all, the data frame is wrong.

But after I made a restriction on the connection, I have filtered the exception tcp connection and kept only my modbus device connected, 
which is the same packet CRC check error, even though the data frame reply is correct every time.
Just like the picture below

All I guess is that after receiving some kind of abnormal data frame, it triggers a bug in the CRC always check error under certain conditions.
And if I reinitialize the acquisition process at this point, then everything is fine.
Just like the picture below

steveohara, 2021-08-20:
I'm going to close this on the basis that there is a proper workaround for re-establishing a connection and I haven't got any further reports of any issues

closed 2021-08-20 by steveohara