steveohara/j2mod #104: Using library with Bluetooth connection

retuner87, 2020-03-12:
Hi all, I'm an android developer and I need to connect my app with a remote bluetooth device that use Modbus RTU protocol; application must request some data and display them to a view.

I create a communication socket with the device but I can not understand how use j2mod lib to send a modbus rtu request.

Can anyone help me?

Thanks in advance
Mattia

steveohara, 2020-03-12:
The library expects to connect to a Morbus slave either via a serial port or a TCP address. If your BT slave manifests it's Comms interface as either of these then it will work.
If not, then you will need to provide your own transport or serial port implementation.
Most people use master facade classes but if you look at those implementations you'll see that they are just helper classes that aggregate the connection, transport and transaction classes.

retuner87, 2020-03-13:
Thanks @steveohara for your reply. This app is the third that I made with a BT connection and I'm not so expert on it. I saw that the UUID connection simulates a serial connection but I can't understand how use it with the j2mod lib, which classes I can use to connect it or which commands.

steveohara, 2020-03-13:
The best way to see how to do this is to follow the example set out in [SerialMaster ](https://github.com/steveohara/j2mod/blob/development/src/main/java/com/ghgande/j2mod/modbus/util/SerialMaster.java)

This is actually a mimic of [ModPoll ](https://www.modbusdriver.com/modpoll.html)and means you can use the j2mod library from the command line to test your serial connections

For example; `java -cp libs/* com.ghgande.j2mod.modbus.util.SerialMaster -a 49 -t 3 -r 1 -l 30 /dev/ttyNS0`

Make sure you have the required jar files (jserialcomm and slf4j) in the `libs` directory

closed 2020-08-07 by steveohara