steveohara/j2mod #106: Holding register function not working as per expectation in rtu mode of modbus slave

DhruvilDhanani, 2020-06-18:
## Expected Behavior
All the request from master is served by serial slave without timeout
## Actual Behavior
Some of the request of master is not served by the server.
This issue is occur if starting address of holding  register is **0**. Other than 0 address slave functionality working as per expectation. 

## Steps to Reproduce the Problem

  1.  Create the serial slave using the library 
  2.  Add the process image of holding register in serial slave
  3. Read data from serial master 

## Specifications

  - Version: 2.6.0
  - Platform: Any
  - Subsystem:

Modbus master frame:

[RTU]>Tx > 13:59:29:162 - 01  03  00  00  00  05  85  C9  
[RTU]>Rx > 13:59:29:225 - 01  03  0A  00  0B  00  16  00  21  00  2C  00  37  1C  5F  
[RTU]>Tx > 13:59:30:570 - 01  03  00  00  00  05  85  C9  
**Sys > 13:59:31:072 - Error : Timeout**
[RTU]>Tx > 13:59:41:122 - 01  03  00  00  00  05  85  C9  
**Sys > 13:59:41:623 - Error : Timeout**
[RTU]>Tx > 13:59:42:288 - 01  03  00  00  00  05  85  C9  
[RTU]>Rx > 13:59:42:372 - 01  03  0A  00  0B  00  16  00  21  00  2C  00  37  1C  5F  
[RTU]>Tx > 13:59:44:306 - 01  03  00  00  00  05  85  C9  
[RTU]>Rx > 13:59:44:389 - 01  03  0A  00  0B  00  16  00  21  00  2C  00  37  1C  5F 

Modbus slave log by debugging 
2020-03-23 08:32:31,295 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.ModbusRTUTransport - Request: 01 03 00 00 00 05 85 C9
2020-03-23 08:32:31,296 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:31,297 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=0
2020-03-23 08:32:31,298 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:31,299 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=1
2020-03-23 08:32:31,300 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:31,301 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=2
2020-03-23 08:32:31,302 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:31,303 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=3
2020-03-23 08:32:31,304 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:31,305 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=4
2020-03-23 08:32:31,305 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:31,306 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=5
2020-03-23 08:32:31,307 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.n.AbstractModbusListener - Request:01 03 00 00 00 05
2020-03-23 08:32:31,309 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.n.AbstractModbusListener - Response:01 03 0A 00 0B 00 16 00 21 00 2C 00 37
2020-03-23 08:32:31,311 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.ModbusRTUTransport - Sent: 01 03 0A 00 0B 00 16 00 21 00 2C 00 37 1C 5F
2020-03-23 08:32:32,295 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.ModbusRTUTransport - Request: 01 03 00 00 00 05 85 C9
2020-03-23 08:32:32,297 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:32,298 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=0
2020-03-23 08:32:32,299 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:32,299 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=1
2020-03-23 08:32:32,300 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:32,301 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=2
2020-03-23 08:32:32,302 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:32,302 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=3
2020-03-23 08:32:32,303 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:32,304 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=4
2020-03-23 08:32:32,305 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:32,306 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=5
2020-03-23 08:32:32,307 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.n.AbstractModbusListener - Request:01 03 00 00 00 05
2020-03-23 08:32:32,309 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.n.AbstractModbusListener - Response:01 03 0A 00 0B 00 16 00 21 00 2C 00 37
2020-03-23 08:32:32,311 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.ModbusRTUTransport - Sent: 01 03 0A 00 0B 00 16 00 21 00 2C 00 37 1C 5F
2020-03-23 08:32:33,294 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.ModbusRTUTransport - Request: 01 03 00 00 00 05 85 C9
2020-03-23 08:32:33,295 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:33,296 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=0
2020-03-23 08:32:33,297 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:33,298 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=1
2020-03-23 08:32:33,299 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
[... 47 lines cut ...]
2020-03-23 08:32:37,322 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:37,322 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=4
2020-03-23 08:32:37,323 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:37,324 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=5
2020-03-23 08:32:37,325 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.n.AbstractModbusListener - Request:01 03 00 00 00 05
2020-03-23 08:32:37,340 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.n.AbstractModbusListener - Response:01 03 0A 00 0B 00 16 00 21 00 2C 00 37
2020-03-23 08:32:37,341 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.ModbusRTUTransport - Sent: 01 03 0A 00 0B 00 16 00 21 00 2C 00 37 1C 5F
2020-03-23 08:32:38,308 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.ModbusRTUTransport - Request: 01 03 00 00 00 05 85 C9
2020-03-23 08:32:38,310 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:38,311 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=0
2020-03-23 08:32:38,311 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:38,312 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=1
2020-03-23 08:32:38,313 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:38,324 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=2
2020-03-23 08:32:38,327 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:38,329 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=3
2020-03-23 08:32:38,330 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
2020-03-23 08:32:38,331 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - count=6 pos=4
2020-03-23 08:32:38,332 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.FastByteArrayInputStream - read()
[... cut ...]
2020-03-23 08:32:38,339 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.n.AbstractModbusListener - Request:01 03 00 00 00 05
2020-03-23 08:32:38,339 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.n.AbstractModbusListener - Response:01 03 0A 00 0B 00 16 00 21 00 2C 00 37
2020-03-23 08:32:38,340 [Modbus Serial Listener [port:/dev/ttyS1]] DEBUG c.g.j.m.i.ModbusRTUTransport - Sent: 01 03 0A 00 0B 00 16 00 21 00 2C 00 37 1C 5F

P.S: Input register functionality is working as per expectation with starting address 0.

steveohara, 2020-06-23:
I think this is working as expected. 
j2mod addresses use the Modbus Number policy which is one based. 
So the first coil/register number is 1, not zero

DhruvilDhanani, 2020-07-02:
Issue is in the serial communication frame. If data frame is valid then application working as per expected. So, Closing this issue.

closed 2020-07-02 by DhruvilDhanani