steveohara/j2mod #109: Unable to read status (register) from the device

parthibunn, 2020-08-05:
ModbusInput modbusInput = new ModbusInput(ipAddress, BARRIER_PORT);
        ModBusService modBusService = new ModBusService(modbusInput);
        //Get the barrier operation based on the request type UP | DN | FR | UF
        BarrierOperation barrierOperation = modBusService.getBarrierOperation("UP");
        //Send request to barrier to update the barrier operation
        boolean isSuccess = modBusService.sendOperation(barrierOperation);
        LOG.info("Barrier status - " + isSuccess);

public boolean sendOperation(BarrierOperation barrierOperation) {
        boolean result=false;
        try {
            modbusInput.setQuantity(barrierOperation.getCode());
            System.out.println("BS Status before - " + iModbusExecutor.readHoldingRegister(modbusInput));
            result = iModbusExecutor.writeSingleRegister(modbusInput);
            System.out.println("BS Status after - " + iModbusExecutor.readHoldingRegister(modbusInput));
            return result;
        } catch (Exception ex) {
            log.error("Exception at sendOperation : ", ex);
            return result;
        }
    }

public boolean readHoldingRegister(ModbusInput modbusInput) throws Exception {
        ModbusTCPMaster master = getInstance(modbusInput.getIpAddress(), modbusInput.getPort());
        try {
            master.connect();
            InputRegister[] inputRegisters = master.readMultipleRegisters(modbusInput.getRegisterAddress(), 8);
            for(int i=0;i<inputRegisters.length;i++) {
                log.info("readHolding - " + inputRegisters[i].getValue());
                log.info("readHolding - " + inputRegisters[i].toBytes().toString());
            }
            master.disconnect();
            return true;
        } catch (Exception e) {
            log.info("Cannot connect to slave - " + e.getMessage());
            return false;
        }
    }

Basically trying to find the change in status/register value before and after write operation. But the values printed are the same

I must be able to retrieve the device status at any point

Any help would be greatly appreciated. Thanks in advance

steveohara, 2020-08-07:
I can't see that you're doing anything wrong here but then I don't know the intricacies of your Barrier controller.
I nearly all cases, this is an issue with your slave and the way it expects you to write to a register and then read it back.
There are plenty of examples in the tests of doing this if you want to take a look there.

parthibunn, 2020-08-11:
@steveohara Thanks for your response. Yes. There is nothing wrong in the code. It was my wrong understanding that caused the issue

closed 2020-08-11 by parthibunn