steveohara/j2mod #112: Problem with raspbian RS-485 communication

pan-henryk, 2020-11-04:
I am using j2mod with jserialcomm on raspberry pi/raspbian derived industrial embedded device.
With jSerialComm-2.4.2 and j2mod-2.5.3 everything works fine, but any higher version it does not.
I noticed that newer version reverts internal com port mode from RS-485 back to rs-232.
Looking at COM port name (/dev/ttySC0) I guest the underlying chip is SC16IS***
Am I doing something wrong or is it a bug with j2mod / jserialcomm?

## Expected Behavior
Modbus communication should work

## Actual Behavior
Modbus communication does not work

## Steps to Reproduce the Problem

  1. Try to establish modbus communication on SC16IS*** chip (device?) running in RS-485 mode, using j2mod version > 2.5.3 and jserialcom version > 2.4.2

## Specifications

  - Version: 2.6.0 or newer
  - Platform: Linux
  - Subsystem:
OS version:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

steveohara, 2020-11-05:
There isn't anything chip specific in j2mod, so I suspect that if the behaviour has changed, then it will be in jserialcomm or perhaps in the linux serial driver that is being used on Raspbian. I would reach out to Will over there https://fazecast.github.io/jSerialComm/ and see what he says.

steveohara, 2020-11-05:
I've re-opened this because I think we can mitigate your problem by exposing the RTS control from jserialcomm through j2mod so that you can control it from your application. 
If you're in a hurry, you could do this yourself by providing your own `AbstractSerialConnection` implementation that extends the included version `SerialConnection` and then setting the RS485 values on Open

pan-henryk, 2020-11-10:
Thank you, I will check it asap.

icedaq, 2020-12-22:
@pan-henryk We are experiencing the exact same problem. Did you ever find a solution for this?

pan-henryk, 2020-12-22:
> @pan-henryk We are experiencing the exact same problem. Did you ever find a solution for this?
I stopped at using last working version (jSerialComm-2.5.1 and j2mod-2.6.4) and got distracted by another project.
I plan to revisit it early January to try solution suggested by Steve.

icedaq, 2020-12-22:
> > @pan-henryk We are experiencing the exact same problem. Did you ever find a solution for this?
> > I stopped at using last working version (jSerialComm-2.5.1 and j2mod-2.6.4) and got distracted by another project.
> > I plan to revisit it early January and try solution suggested by Steve.

Thanks for the update. We tested with the version you mentioned and it works perfectly in our case too. So we will stick with this version for the moment.

steveohara, 2021-08-20:
Closing this on the basis that https://github.com/steveohara/j2mod/pull/117 has now introduced RS485 mode which should fix the issue

closed 2021-08-20 by steveohara