steveohara/j2mod #113: How to add special function in ModbusMaster

Julien-Faure, 2020-12-07:
I would to add special function in ModbusMaster, can you help me please ?

Thank you in advance.

steveohara, 2020-12-10:
What is the special function you want to add?
You could of course just clone, add the code and create an MR if you're confident it will be good for the library

Julien-Faure, 2020-12-11:
I wanted to know how to add custom function.

Read registers or write registers value are official functions.

### For example :

I want to create a function with id 80.

This function could reboot a remote slave.

steveohara, 2021-08-20:
ModbusMaster is an aggregate of other public classes that can do this if you use them directly. I would like to say that I can provide this as a feature of j2mod out of the box but I don't have time I'm afraid.  Take a look at some of the examples under com.ghgande.j2mod.modbus.cmd for more concrete help

closed 2021-08-20 by steveohara