steveohara/j2mod #119: How to get response when I write a command using SerialConnection writeBytes() method

aviral1999099, 2021-08-18:
## Expected Behavior
The readBytes() method should return a byte array 

## Actual Behavior
but it only returns int value and maximum times its length is 0

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Version: 2.7.0
  - Platform: Linux
  - Subsystem:

aviral1999099, 2021-08-18:
I could not find a method to read a particular register, can you help me here. This is my ultimate task.

steveohara, 2021-08-20:
I'm not sure what you're asking, seems kind of basic i.e. "how do I read a register using j2mod?"
If that is the case, then please refer to the myriad of examples in the tests or look at the documentation https://github.com/steveohara/j2mod/wiki/Using-j2mod-to-read-from-a-Slave#read-input-registers

closed 2021-08-20 by steveohara

aviral1999099, 2021-08-20:
Thanks for your help. I actually did not phrased the question right. I just
wanted to read register i.e. the value they are holding. I did find it by
the way.