steveohara/j2mod #125: Cannot open virtual one specific COM port on Windows

pan-henryk, 2022-02-04:
## Expected Behavior
1. Port opened normally via j2mod
2. The same port port opened normally via old javax.comm serial library

## Actual Behavior
1. j2mod throws exception
java.io.IOException: Port [COM55] cannot be opened after [3] attempts - valid ports are: [COM1,COM11,COM12,COM13,COM14,COM15,COM16,COM2,COM21,COM22,COM3,COM30,COM32,COM33,COM35,COM36,COM37,COM38,COM39,COM4,COM40,COM41,COM45,COM46,COM47,COM5,COM55,COM6,COM7,COM8,COM9]
	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:135) ~[j2mod-3.0.0.jar:3.0.0]
	
2.  The same port port opened normally via old javax.comm serial library	

## Steps to Reproduce the Problem
Honestly I have no idea what is going on, looking for any clues.
I have a virtual COM port that opens and communicates normally via legacy javax.comm driver, but fails to open via j2mod/jserialcomm. I have tried several different version combinations ranging from quite old to latest and all produce the same effect.
Whats weird - it is a first port of 2-port MOXA 5230 device. The other one, configured exactly the same opens without any problems. 
The only unusual thing I see is that this particular device is configured for 2 stop bits, while most of the others have 1 stop bit.

## Specifications

  - Version: Windows Server 2019, x64 (virtual machine) and  Windows 2008 R2 Standard SP1 (also virtual machine) verified on two different machines.
  - Platform:
  - Subsystem:

steveohara, 2023-09-14:
I can't replicate this on the latest version 3.2.0
Can you try this again and report back

steveohara, 2023-10-21:
The opening of serial comms ports is handled by jserialcomm so you will have better luck talking to those guys

closed 2023-10-21 by steveohara

pan-henryk, 2023-10-30:
I think I got it :)
Serial Parameters doc says stopBits passed as int means "number of stopbits" while in fact it should be a reference to 
jserial comm constant defined as following:
ONE_STOP_BIT= 1
ONE_POINT_FIVE_STOP_BITS = 2
TWO_STOP_BITS = 3
So I was passing 2 as an int and it meant something different than I thought. Maybe you could update javadoc to reflect this?