steveohara/j2mod #126: Move parts of DEBUG logging to TRACE

gitMiguel, 2022-02-06:
While developing a modbus tool based on this library and switching loggin to debug I noticed these lines repeating quite often. IMO they should go to TRACE. Is there a reason why they are not? You think they could they be moved? 

## Example log

> 2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - reset()
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - mark=0 pos=0
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - mark()
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - mark=0 pos=0
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=7
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - reset()
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - mark=0 pos=0
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,516 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=0
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=1
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=2
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=3
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=4
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=5
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=6
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=7
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=8
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - read()
2022-02-06 09:37:00,517 [DEBUG] [.FastByteArrayInputStream] - count=10 pos=9

> 2022-02-06 09:37:00,517 [DEBUG] [mod.modbus.util.BitVector] - Get bit #0

LivingWithHippos, 2022-02-08:
How did you set up the log level?

gitMiguel, 2022-02-13:
I'm developing with eclipse and use log4j. I have defined the logger like this:
`<Logger name="com.ghgande.j2mod.modbus" level="debug"/>`.

steveohara, 2023-09-13:
Agreed - changed in next build

closed 2023-09-13 by steveohara