steveohara/j2mod #132: Unexpected Connection Closed by peer

chaos84731, 2022-08-31:
## Expected Behavior
Being able to write to coils and registers (single and multiple) without the connection being closed.

## Actual Behavior
Random closure of the connection after writing to coils or registers.

## Steps to Reproduce the Problem

  1. Send command via Simply Modbus TCP Client
[image]

  1. Have the connection randomly close.
  1.

## Specifications

  - Version:  j2mod:3.0.0'
  - Platform: Android

Hi Randomly I will get a Error 66 (connection closed by peer)  when I am writing registers.  Please help.

Thanks

steveohara, 2023-09-13:
Is it possible you can reproduce with a unit test?

chaos84731, 2023-10-18:
Turns out the Error was in my code and updating the Slave while writing values.

closed 2023-10-18 by chaos84731