steveohara/j2mod #135: Add capability to bind the Socket to specific local interface.

tamiradler, 2022-12-25:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Version:
  - Platform:
  - Subsystem:

steveohara, 2023-10-21:
Can you explain the use case you have for this?
I'm struggling to understand why you wouldn't want to just rely on the OS to select the correct adapter for an outbound connection. The OS will figure out which network card to use based on your destination address and for that it uses routing tables etc.

steveohara, 2023-12-09:
It was easy enough to add but tricky to test so there are no unit tests for it.
I would appreciate it if you could take the snapshot for a spin and let me know if it meets your needs.
New getter/setter added to `TCPMasterConnection` to allow the user to select a network interface to bind to.

closed 2023-12-09 by steveohara