steveohara/j2mod #136: Request for quick example update for logging

dannaughton, 2023-01-09:
Reqeust for simple update to documentation or example to control log levels.

In on of the example:
private static final ModbusLogger logger = ModbusLogger.getLogger(Test.class);

ModbusLogger does not exist.

My modbus slave I created is just filled with debug information - I don't know how to throttle it.  Thanks

mkurt, 2023-01-09:
Hi,
You can change log levels by configuring your logger implementation. If you want less verbosity you can set level to info or higher. Here is an logback config example from my repo: [Logback config](https://github.com/inscada-app/ins-modbus-cli/blob/3768c9492b673dbf9ae04f8bf96a2c511f774821/src/main/resources/logback.xml)

steveohara, 2023-09-14:
j2mod uses [sl4j ](https://www.slf4j.org/) which is a façade on top of whatever your favourite logger is that you're using in your application. As mkurt suggested, [logback](https://logback.qos.ch/manual/configuration.html) is a good one, so is [log4j2](https://logging.apache.org/log4j/2.x/manual/configuration.html) and there are others. That is where you would manage the log settings.

closed 2023-09-14 by steveohara