steveohara/j2mod #142: Misleading documentation for SerialParameters

pan-henryk, 2023-10-30:
This refers to a bug I had a few months ago:
https://github.com/steveohara/j2mod/issues/125

I think I finally understand what is going on.
Serial Parameters doc (constructor and setStopbits functions) says stopBits passed as int means "number of stop bits" while in fact it should be a reference to jserial comm constant defined as following:
ONE_STOP_BIT= 1
ONE_POINT_FIVE_STOP_BITS = 2
TWO_STOP_BITS = 3 (!)
So I was passing 2 as an int and it meant something different than I thought. Maybe you could update javadoc to reflect this?

pan-henryk, 2023-10-31:
I created a PR with proposed changes:
https://github.com/steveohara/j2mod/pull/143

steveohara, 2023-11-10:
Thanks pan_henryk

closed 2023-11-10 by steveohara